16n-faderbank/16n #4: Add version number to Eagle files?

okyeron, 2019-01-03:
At first glance it's not obvious which eagle files are the ones to use.

Adding a version number or putting the old (1.26) version in a separate folder would help reduce confusion.

infovore, 2019-01-03:
v1.26 has been removed from the repository, because we have git history now, and no-one really should be using 1.2x.

My goal is to only have one current version. Version number is written on the rear silk of the BoaRD file - it should probably be in the SCHematic, too, though. Thoughts?

infovore, 2019-01-03:
I thought about removing v1.26  from the repository, because we have git history now, and no-one really should be using 1.2x. However, I might bung it in a folder called `legacy` and leave it at that.

My goal is to only have one current version. Version number is written on the rear silk of the BoaRD file - it should probably be in the SCHematic, too, though. Thoughts?

infovore, 2019-01-03:
(See 
https://github.com/16n-faderbank/16n/tree/master/electronics/legacy)

okyeron, 2019-01-03:
Adding version to the schematic would be great.

closed 2019-01-15 by infovore