16n-faderbank/16n #7: Standoff height?

okyeron, 2019-01-03:
Another discrepancy between Octopart BOM and CSV...

The CSV has 6mm M-F standoff, Octopart has 5mm 

Should we assume Octopart BOM is correct?

infovore, 2019-01-03:
Well.

6mm is correct, but even though Harwin make a 6mm M-F spacer, Octopart doesn't list it.

Also: this is dependent on the length of your screws. With the counterrsunk screws, I needed a 6mm spacer to accommodate them; with panheads, the 5mm one was fine. I could go looking for alternate parts. If you're not using countersunk holes, 5mm should be fine.

This is a weird discrepancy, but I'm not sure how best to solve it.

okyeron, 2019-01-03:
So here's alternate part from mouser that's 6mm, but only 4.5 wide
https://www.mouser.com/ProductDetail/RAF-Electronic-Hardware/M2101-2545-AL?qs=sGAEpiMZZMtrde5aJd3qwxLSm3K1%252btbtLUSkuQwkkwQ%3d

Might be good to mention the length being an option in the docs?

okyeron, 2019-01-28:
Followup thought -  if your fader bats/arms/shafts (what are those called anyway) are too long/tall for your chosen fader caps, you could potentially get longer top spacers to raise the panel up a bit.

Useful addendum  for the build docs?