16n-faderbank/16n #9: Issue compiling on Windows

infovore, 2019-01-10:
As seen here: 
https://llllllll.co/t/16n-is-a-bank-of-faders-release-thread-current-version-1-31/18620/118

> I’m having problems compiling the code. 1.8.8/1.45 on win10 64bit.
> `_16n_faderbank_firmware:27: error: expected constructor, destructor, or type conversion before ‘;’ token MIDI_CREATE_DEFAULT_INSTANCE();`

altitude909, 2019-01-10:
following.

infovore, 2019-01-10:
@altitude909 so here's a question - in Sketch -> Libraries -> Manage Libraries  - can you find "MIDI Library by Forty-Seven Effects" and tell me what version you have installed? You shouldn't need this - Teensyduino includes a midi library - but I'm wondering if there's a version conflict. (I have 4.3.1 of this library installed, and all compiles fine).

altitude909, 2019-01-10:
yeah, wouldn't surprise me I think im up to 7 midi libraries in there. its still up on screen, I'll check in a bit

okyeron, 2019-01-10:
I think the 47-effects lib is deprecated now (after 1.8.5). The newest teensyduino should override this.

You could try disabling the 47-effects lib and/or reinstall teensyduino

infovore, 2019-01-10:
@okyeron in thread, @altitude909 said they were on Arduino 1.8.8, Teensyduino 1.45 (ie, all latest)

okyeron, 2019-01-10:
Right - scanner_darkly had the exact same error last night on some other teensy/midi code and he fixed it by upgrading to 1.8.8 and 1.45 (on windows).

In this case my guess is the 47effects midi lib is getting loaded instead of the Teensy one?

infovore, 2019-01-10:
@okyeron no; I have the 47effects library installed; everything compiles fine. However - I was wondering if there was a conflict with _older_ versions of 47effects, which had similar issues to the one described.

altitude909, 2019-01-10:
Ok, Used: p:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MIDI is what worked.  I had a who knows what lib that is was trying and that was breaking the build

infovore, 2019-01-10:
@altitude909 ah, yeah - so there were some changes to the core MIDI libraries a while back; older versions seemed to trigger the error you described. Old things kicking around in `.../libraries` might do that.

I'm glad you've got it working. Outputs from this: I'm going to add this clarification in the `README` before closing this issue.

closed 2019-01-18 by infovore