16n-faderbank/16n #12: I2C connection issues in slave mode.

Elberstein, 2019-01-11:
I'm experiencing issues with the I2C connection. I'm testing the connection with a Teletype, when polled, it always returns 0. (See "Context" below) 

I'm not sure wether it's related to hardware or software. If it's related to the hardware, the only change I made is replacing the 1N4148W-7-F diodes by 1N4148W**Q**-7-F diodes. The specs are nearly identical but the Q-7-F have a reverse current of 50uF instead of 1uF for the 7-F.
Could this be the cause of the issue ? 

**Context:** 
- Teletype (old hardware, not the latest revision) Firmware number -> 25E14EE
- TT busboard.
- 16nFaderbank in slave mode
- 16n connected to the I2C bus through a TXo.
- Total cable length outside + inside the eurorack case: approx. 35cm
- Connectivity between the Teletype and the Teensy tested with a multimeter: OK (Tip: SDA, Ring: SCL, Sleeve: ground. Pins 18 and 19 on the Teensy)
- The firmware of the 16n has not been modified (except for enabling DEBUG in config.h)

Other I2C devices on the bus: 
Backpack header 1: ER-301
Backpack header 2: Two TXo + one 16n connected in series.

**Situation:** 
- The ER-301 and the TXOs receive messages from the Teletype. 
- The 16n in DEBUG mode doesn't receive messages from the Teletype according to the Serial Monitor. 
- When polled from the Teletype with `FB 1`, the result is always 0, same thing for each fader.

Thanks.

Elberstein, 2019-01-12:
New test: Faderbank directly connected to the Teletype via i2C, removed all the other devices on the bus.
Result: Apparently the 16n stopped working (judging by the USB test in browser and the LED not lit on the Teensy I had to re-upload the firmware.

Elberstein, 2019-01-13:
Thanks for your reply. 

I've just made a new test connecting the 16nFaderbank directly to the Teletype without busboard this time. As I wrote yesterday, the 16n didn't like being the only device on the I2C bus with the pull-up resistors of the TT busboard. This time, the unit doesn't stop working but it always returns 0. 
I have to say that all the tests have been done with two 16nFaderbank successively, I built two.

Elberstein, 2019-01-14:
In Master mode this time, the serial monitor is full of i2c messages, the 16n is connected to the ER-301
but it doesn't receive anything. I think that's normal since the faderbank seems to send i2C to an address that is not longer used by the ER-301. 

```
      // ER-301
      sendi2c(0xB0, 0, 0x11, q, notShiftyTemp);
```

The new address is 0x31 I'm opening a new issue for that. 

      // ER-301
      sendi2c(0x31, 0, 0x11, q, notShiftyTemp);

[image]

Elberstein, 2019-01-14:
OK, the issue is now fixed @infovore and @bpcmusic , it's also an issue related to the I2C address. 

The address should be **0x34 and not 0x80** according to https://github.com/monome/libavr32/blob/master/src/ii.h

```
  #else
  Wire.begin(I2C_SLAVE, 0x34, I2C_PINS_18_19, I2C_PULLUP_EXT, 400000); 
  Wire.onReceive(i2cWrite);  
  Wire.onRequest(i2cReadRequest);
```

bpcmusic, 2019-01-15:
Looks like there was some drift since the initial versions on both sides (TT and ER-301). I've been so neck-deep in my build that I haven't kept up. Thanks for catching it!

Elberstein, 2019-01-15:
Yes, for me the changes you've made are OK @infovore

infovore, 2019-01-18:
`master` branch is now up-to-date with correct addresses.

closed 2019-01-18 by infovore