16n-faderbank/16n #14: Flip device

Elberstein, 2019-01-15:
It would useful to have the possibility to flip the device, I2C connector on top, midi connector on the left.

In config.h there's an option for the fader numbers `#define REV 1` but what is the best way to also invert the values ? 

I replaced   `temp = map(temp, MINFADER, MAXFADER, 0, 16383);`
with   ` temp = map(temp, MINFADER, MAXFADER, 16383, 0);`

It works but the lowest value is always 2 or 3. How can I adjust that please ?

infovore, 2019-01-15:
I'll have a think. I think the nice feature would be to have a single `define` flag like the L-R reversing, called `FLIP`, that would just make this happen.

infovore, 2019-01-15:
In the meantime - have a go at reducing `MINFADER`, which is line 34 in `config.h` - basically, might be worth fettling the sensitivity of the analog read. I'll look into this further in coming days. For now, I'm just kicking off a branch to work on this.

Elberstein, 2019-01-15:
Unfortunately, the hack `temp = map(temp, MINFADER, MAXFADER, 16383, 0);` doesn't seem to work with I2C, and reducing the value of MINFADER does't seem to have any effect

infovore, 2019-01-15:
OK, duly noted. Pretty sure I can come up with a solution for this - but not before Friday this week!

infovore, 2019-01-18:
OK, @Elberstein . I've taken a quick pass at this in a branch and got a PR open for it - https://github.com/16n-faderbank/16n/pull/17 . If you're comfortable downloading that branch and flashing that firmware, can you let me know how you get on? I've tested it on my 16n, which, over USB MIDI, behaves correctly when `REV` and `FLIP` config settings are set to 1. We can move discussion over to the PR if you'd like.

Elberstein, 2019-01-18:
I'm not really familiar with Github, I have to learn but it's not a problem. 
Thanks for adding this feature. I've just tested it with the Teletype. It works but the minimum value is around 350 even if MINFADER is set to 0

infovore, 2019-01-18:
OK, I think I have it working now. If you get the firmware in this branch - https://github.com/16n-faderbank/16n/tree/flip_feature - and enable `FLIP` and `REV`, my debug log seems to send correct I2C out when the thing is upside down.

Elberstein, 2019-01-18:
Yes it works but I have to increase the MINFADER value up to 200

infovore, 2019-01-20:
Merged to master.

closed 2019-01-20 by infovore