16n-faderbank/16n #16: Multiple faderbanks on the same I2C bus

Elberstein, 2019-01-16:
Is it possible to have multiple faderbanks on the same I2C bus ? I'm wondering if it has already been tested ?

Out of curiosity I tried to add an offset here but of course, it doesn't work...

```
    // map and update the value 
    noInterrupts();
    currentValue[i] = temp;
    interrupts();
```

Thanks.

infovore, 2019-01-18:
I don't think so. Disclaimer: I am not the I2C expert.

Basically, in non-master mode - the default for Teletype - each device needs an address, and Teletype only knows one 16n address (for the device `fader`); adding more faderbanks would need more addresses over in Teletype land.

It _might_ be more possible in MASTER mode, because it's multiple devices sending to endpoints, but even then, code modification would be required - setting custom addresses, possibly changing the fader data sent over I2C.

So: for now, I'm paging Brendon (when he has time), but my gut says not without significant work. I also think that work would be largely user-unfriendly for an edge case.

Elberstein, 2019-01-18:
Thanks for your reply.
Yes, I see two options for the Slave mode:

A: Probably the best solution. More addresses in Teletype like you're suggesting (i.e: 0x35 etc.) and the user can change this line to reflect the address of the device   `Wire.begin(I2C_MASTER, **0x35**, I2C_PINS_18_19, I2C_PULLUP_EXT, 400000); ` or a variable in config.h (?)

B: Increasing the number of ports in Teletype (fader) and adding an offset (+ 16, +32) to the port number, that seems more complicated. 

In both cases, it requires a modification of the teletype code.

Multiple Masters on the same bus: I'm not sure what will happen but I need to try that, if one faderbank can talk to the Teletype and another to the ER-301 while keeping the ER-301 connected to Teletype, that would be great.

bpcmusic, 2019-01-19:
You could have a one in normal mode and one in master mode, i think (but haven't tried it). Things get complicated with the Teletype as it isn't a fan of multiple masters at the moment. So, until that gets resolved (and scanner_darkly has a lead on that), you wouldn't have a stable multi-master configuration.

If you want to have two that you READ from, we could enhance the firmware to support another set of addresses. It would work the same way that it was coded for the TXi. Basically, the Teletype knows that each device has N inputs and for queries outside of that index, it advances in the address space one and normalizes to the appropriate input on the device. For example, TI.PRM 5 = TI address #2; knob 1.

That all means adding more sequential addresses for the 16n and updating the TT code to know about multiple 16n. At the bare minimum, the 16n firmware would need to be updated and compiled with the new address(es). It also could be modified to increment the ports that it sends to in Master mode - which would make sense.

I decided not to do all this (even though I have the code from my TXi implementation) to keep things simple and easy to modify by others. Given the Teletype's script limitations, I've found creatively using all 16 sliders to be a challenge in and of itself. :)

infovore, 2019-01-20:
@Elberstein I think this is effectively resolved? Before I close it, I think the best thing to do is convert @bpcmusic's response into some documentation. Is that OK?

Elberstein, 2019-01-20:
Yes, the work is on the Teletype side, I think we can close this issue :-)

closed 2019-01-20 by infovore