16n-faderbank/16n #19: ansible 12c address mismatch?

okyeron, 2019-01-28:
small confusion...

[ii.h](https://github.com/monome/libavr32/blob/master/src/ii.h) has 4 ansible addresses listed - and none of those match the ansible address in the 16n code

ii.h:
```
#define II_ANSIBLE_ADDR     0x20
#define II_ANSIBLE_ADDR_2   0x21
#define II_ANSIBLE_ADDR_3   0x22
#define II_ANSIBLE_ADDR_4   0x23
```

16n:
```
      // ANSIBLE
      sendi2c(0xA0, device << 1, 0x06, port, notShiftyTemp);
```

infovore, 2019-01-28:
Hmn, interesting. I'd assume it should be `0x20`, but I think @bpcmusic is your best bet for the right answer here.

scanner-darkly, 2019-01-28:
we use 7-bit addresses, so the MSB will get ignored, which makes address 0x20 equivalent to 0xA0. still, probably best to use the addresses defined in `ii.h` to avoid confusion (and for monome ecosystem we should use `ii.h` as the source of truth).