16n-faderbank/16n #24: Panel thickness?

cschol, 2019-02-16:
Using the Alps faders from the BOM, what is the optimal panel **thickness**, especially for the top panel?

okyeron, 2019-02-20:
Depends on your fader caps maybe? 

The mixer style caps I have are meant to be more flush to the panel so I found using a 13mm spacer with a 1/8in panel works well.

infovore, 2019-02-20:
In the images you've seen of the aluminium panel, I've been using 2mm aluminium. I've also successfully made one with 3mm plastic. The plastic one uses 10mm spacers up top, 5mm below; the metal one uses 6mm spacers below owing to countersunk holes requiring a bit more depth for the screws. Does that help?

cschol, 2019-02-20:
Yes. Thanks! I have ordered a 3mm plastic panels. That should work.

closed 2019-02-21 by infovore