16n-faderbank/16n #26: faders not going to full value on midi

altitude909, 2019-04-17:
Someone brought up that their device is sending CC values over 124 and I checked several devices here and some do some do some dont.  Its persistent over all the faders for given device.  5V is good in all cases..

jamiemccusker, 2019-05-06:
Same issue here in the one I’ve built . Midi goes from 0-124. Any ideas?

andrewh, 2019-05-06:
I've also had issues finding the right values for `MINFADER` and `MAXFADER` in `config.h`. Would be good to know what others are using to get stable 0-127 over usb-midi.

I've played around and had `MINFADER` at 294 which went to MIDI 0, and `MAXFADER` at 8135 which went to MIDI 127, but the Arduino app's serial debug monitor picked up values coming in from other faders (all set to 0), so maybe there is a sensitivity issue for certain values in `config.h`?

pppier, 2019-05-25:
I also read 26 to 100 picking midi from TRS; and 0 to 124 using MIDI over USB... 
I hope there's a way to have 0-127 on both...
is there a  way to check if I2C is working even if not owning a monome (like the midi html test page...) ?

jonnybutler, 2019-11-15:
I’ve noticed this happening depending how it’s powered. I had it plugged into my computer via a hub and that was enough to lesson the voltage it was operating with, which I think in turn affected the max midi value.

altitude909, 2019-11-15:
That's how it works.  The reference voltage for the faders is directly taken from the USB supply voltage and those can vary widely.  What ive been doing is lowering the range of the ADC to 8000, this gives a bit of a dead spot at the top some times but it will always go to 100%

andrewh, 2019-11-19:
So maybe the fix here is just some documentation?

scaleandthreshold, 2021-08-16:
All my faders go from 0-125, I confirmed this in the MIDI Test Page.
I then installed the latest firmware meaning I can't use the MIDI Test, but can now use the Configuration Tool. 
Same problem.

I'm not using CV just MIDI over USB.