16n-faderbank/16n #29: Motorised Sliders

deskofcraig, 2019-12-16:
Is it possible to use motorised sliders that can snap to preset values from the host to give a true value when using the 16n?

infovore, 2020-01-20:
Right now, it is not "possible" in the sense that you can't just put motorised faders in place of the current faders, write some code, and have it work.

Yes, it would be cool.

This idea has been suggested in multiple locations on multiple occasions, and it's worth going over the changes necessary to doing this, if only to archive them:

* the firmware needs to parse MIDI input to take the values the host is sending it.
* the electronics now need to be added to, to send values to the motors - is this analog? is this stepper instructions? regardless, there are extra constraints, and more pins on the Teensy/MCU to be used (and possibly more support componentry to be added. Motor drivers, possibly?)
* the fader parts need replacing with motorised hardware. these are physically larger than the current parts, primarily _depthwise_.
* ...which means that the board will need re-designing, and it's going to be a physically bigger object...
* ...which will mean redesigning the panels/case.

Motorised alps 60mm faders are around... 7x the cost of the standard faders. You'd spend €400 on faders alone to build this.

As you can see, it's not easy, or cost effective. Personally, I'm not very interested in working on this project.

However: 16n is opensource, and you have all the starting points to adjust the design.

sapyatkuen, 2020-10-29:
You can find in Thomann a 5 Behringer motor fader kit for 89€ and there are plenty of other ones in AliExpress. 

In my case I´d be very interested in a 16n with motorised faders to use with Ableton, or at least a motorized 8n (with a button that sends CC61) to connect to the Octatrack.

infovore, 2020-10-29:
@sapyatkuen fine, so you've found some cheaper faders, and a 16n made with Behringer faders would cost you 287 euro, not 400. however, that behringer part is.. an I don't know what, so you need to go find a footprint for it, or make one; it looks alps-ish, fortunately. If you look at the form factor, and where the motor goes, you now have to leave holes in the PCB, which will make it much larger in terms of both the Y axis of the pcb and the Z axis of the enclosure. And then you have to wire in the stepper motors, and add stepper motor drivers, and and and... all my comments around the electronics still stand, and we've only saved 100 euro.

(In general, I like projects where parts are _sourecable_, and relying on a random thing from AliExpress is no guarantee it will be available in future. So that's not really a solution.)

I'm closing this thread, because "motorised faderbank based on the 16n project" is a new project, not a neat feature we can add, and not one I have any interest in adding.

As ever, the 16n mantra applies: _that's a really interesting idea, but Tom's not interested in doing it. The whole project is open source, and you can get all the tools to make these changes. Why not try?_

closed 2020-10-29 by infovore