16n-faderbank/16n #33: Add inbound sysex message for USB options.

infovore, 2020-04-09:
32 byte inbound message for USB-specific options: 16 bytes of channels, 16 bytes of CCs.

Note that this data is not contiguous inside EEPROM, so will need two write passes on the device firmware.

infovore, 2020-04-09:
Resolved in bd058c79f209219ed7a57085bd62af94bf46ee0b.

closed 2020-04-09 by infovore

altitude909, 2020-04-09:
The 60 Byte thing is a quirk of the teensy, you can change it in \avr\cores\teensy3 usbmidi.h to 90

infovore, 2020-04-09:
Yeah, I found that after some late-night googling. I decided, rather than altering core code, to work around it. 

So the current fix is to identify the LC boards (and have them identify themselves to the editor), and make sure that rather than sending them 90 bytes of sysex, we send them 3 lumps of < 60. (In this case, 16, 32, and 32). It wasn't so challenging to sort that out, and makes it more reliable to compile up the firmware from source. It also means we don't have to change that again in the future. The bigger limitation on the LCs is their limited EEPROM size - we're underneath the 128byte limit for sure, but it's a thing to be wary of in the future.

Just testing my supposedly fixed firmware with some US folks who are helping me, and we should have a fully compatible thing out over the weekend. That's the plan, anyhow.

infovore, 2020-04-09:
@altitude909 all changes on the firmware side are in this (in progress) PR: https://github.com/16n-faderbank/16n/pull/36/files