16n-faderbank/16n #37: Slider resistance is absent in BOM

vafu, 2020-04-30:
seems that https://github.com/16n-faderbank/16n/blob/master/electronics/bom.csv doesn't say anything about resistance of sliders needed. also, octopart link gives 404.
what is correct slider resistance value ?

infovore, 2020-05-12:
The part reference for the slider - `RS6011SP6003` is specifically for a 10k linear part. You can also get 20k linear, they'd work fine. The resistance doesn't really matter, but it's important that they are linear, rather than log, taper.

Octopart BOM shouldn't be 404ing. I'll look into that.

altitude909, 2020-05-12:
Just FYI, ALPS doesnt make a linear pot in anything but 10KB.  AFAIK, Alps is the only stocked part in that configuration (single B10K in a dual footprint) out in the wild so it has to be that exact part

infovore, 2020-05-12:
oops, thank you for clarification, @altitude909!