16n-faderbank/16n #38: Only send messages when values change

trentgill, 2020-05-03:
At the moment MIDI and I2C messages are sent as fast as the main loop can.

If a tiny amount of hysteresis was added to the smoothed analog value, messages could be only sent when the sliders are moved. This could make 16n coexist on an i2c bus (as a leader) with multiple leader devices.

Is there any reason why this would be a bad idea? If not, I'm happy to submit a PR to do this.

infovore, 2020-05-28:
I was about to write "hang on, I thought we throttled this' (there's been some rewrites of the analog-read code over time). We appear to be using a mixture of `ResponsiveAnalogRead` and manual "has something changed"; really, RAR provides a lot of that - including IIRC a little hysteresis for us.

So: possibly worth doing a tidyup to use the code we've got better.

But I genuinely thought we _were_ only sending messages when sliders changed? Are we not? Or are i2c and midi behaving differently? Sounds like something fishy is going on here.

trentgill, 2020-05-28:
Full disclosure I don't have a 16n to test with! I had just made this assumption from reading the code and not being familiar with `ResponsiveAnalogRead`'s semantics. Apologies for opening this needlessly! I've just been trying to deal with the context of allowing multiple-leader devices on the same i2c bus (specifically re: Crow).

Closing this issue & thank you for the explanation!

closed 2020-05-28 by trentgill

infovore, 2020-05-30:
TBH it's still all a bit ugly, and could probably be tidied - I hate how messy some of the forking has got. So it's still worth a look.

And - I'm dealing with the mirror problem: I have a 16n but no i2c devices, and so I'm largely in the dark on i2c fixes.