16n-faderbank/16n #39: Teensy troubles

ian-c121, 2020-06-17:
hi all,
apologies if this is a stupid question, but i'm having issues with my teensy.
i'm a kit builder, so the build was nice and straightforward. now "all" i've got to do is shove a bit of code into it.
i've downloaded everything, opened the loader, done the verify (tick) stage, but now i'm stuck.
how do i get the v201 hex file into my teensy??

any help would be much appreciated.

cheers,
ian c.

infovore, 2020-06-23:
When you say, "opened the loader", do you mean "Arduino" or the actual Teensy Loader software?

[The instructions on the wiki are your best bet](https://github.com/16n-faderbank/16n/wiki/Firmware:-installation-instructions); you should download a standalone Teensy Loader, and from that little window you can 'open'/load the .hex file. You should then be able to flash it to the Teensy that's connected. If you're having difficulty with the connection, check your micro-usb cable is a real data cable, not just a charging cable.

You don't need to compile the sourcecode yourself in Arduino.

Sorry if that's just repeating what the wiki says.

ian-c121, 2020-06-23:
hi infovore,

thanks for the message.

i'm referring to the teensyduino. i'm sure i'm ok up to that point, but then the fun starts.
good thought about the cable, though. i'd had that thought myself, as everything else looks ok. so i'll check that and try again.

thanks again,
ian c.