16n-faderbank/16n #40: v201 non-working + weird debug message

modularev, 2020-07-24:
Hi there!

Not sure what the issue could be, but my 16n worked flawlessly with the v134 firmware but is now acting up when installing v201.

I tried installing the hex and compiling it myself. Serial Monitor tells me:

`Got an 1nFo request`
`Sending this data`
`7D 00 00 0F 02 02 00 01 01 4C 41 43 02 00 00 00 74 00 D9 CB D4 13 5E 2D E7 46 75 60 F8 79 80 93 09 AD 91 C6 19 E0 A1 F9 7F 7F 9A 13 29 2D B2 46 3A 60 CC 79 52 93 DF AC 65 C6 EB DF 76 F9 7F 7F 84 13 09 2D 94 46 18 60 A6 79 2B 93 B1 AC 3D C6 BF DF 43 F9 7F 7F D8 13 `
`Ooh, sysex`
`F0 7D 00 00 1F F7`

Thanks a lot and all the best!

infovore, 2020-07-27:
So, if you're looking at serial monitor, and have debug messages enabled, and are watching serial monitor whilst you're using the editor tool, that looks OK in the sense that it's the correct debug code.

However, the data being sent looks like garbage:

`7D 00 00` is our manufacturer message.
`0F` is the indicator we're sending a `c0nFig` message
`02` is our device-type ("16n faderbank")
`02 00 01` is our firmware version (2.0.1)

and then the rest is an eeprom dump, and many of those bytes are just not right.

So, question: did you use this Teensy for anything else before this project? I ask because I think this is related to #41

modularev, 2020-07-27:
I‘m the one from the lines forum so I guess this issue can be closed :)

closed 2020-07-27 by modularev