16n-faderbank/16n #41: v2 Firmware: add force-factory-reset sysex message.

infovore, 2020-07-27:
A 16n with v2.0.x firmware starting up for the first time checks its EEPROM to see if it needs to do a factory initialisation. It does this by looking at the first byte of sort-of-EEPROM; if it's not `0xff`, it assumes it's already been programmed; otherwise, it fills the EEPROM with default settings.

This is a problem if your Teensy has been used in a previous project, and has a non-`0xFF` first byte, but the rest of the EEPROM is not really 16n data; that data gets sent as garbage Sysex.

So:

* Add a new sysex receive message that will force a factory reset.
* Add some kind of way of triggering this from the editor tool, or a new endpoint that can handle it.

modularev, 2020-07-27:
I just ran
` for (int i = 0 ; i < EEPROM.length() ; i++) {`
`EEPROM.write(i, 0);`
`}`

which made it get recognized by the editor, and stopped outputting random messages.
However, now I don't get it to transmit anything via USBMIDI, something I failed to see in my quick solution?

Thanks and all the best!

infovore, 2020-07-28:
@modularev Amazing! That's really good to know because it confirms that this is the issue. I've got the sysex end of things written, just need to work out how to add it to the editor in a sensible way - and then people who aren't editing code can have the same fix.

modularev, 2020-08-19:
A weird thing I noticed with my solution is that the editor won't show any value of the fader when cc is set to sth else than "standard"
e.g. fader 16 with init cc 47 doesn't show it's value in the editor when changed to cc 88
the 16n regardless sends it's the correct value on the correct midi channel...

modularev, 2020-08-19:
nevermind, initializeFactorySettings(); solved it.

infovore, 2020-08-19:
good to know: the change I'm making basically runs `initializeFactorySettings()` when sent a sysex message. should be done in the next few weeks.

PR #42 (2020-12-29): Add functionality to reset EEPROM remotely
This PR reflects work in progress to fix #41: resetting the EEPROM of a Teensy via a sysex command, allowing 16ns built with re-used Teensys to not require any heavy lifting to fix up.
closed 2020-12-29 by infovore