16n-faderbank/16n #46: Fader 1 controlling all values

apossibleworld, 2020-09-05:
Hello -- I recently purchased a used 16n on Reverb.  I just updated to the latest firmware, all good.  In the web editor, fader 1 is controlling all of the other fader values, and the other faders do nothing.  So in the editor display, when I push up the first fader, I see all 16 move up to 127.  

Any suggestions or help?  I don't know who assembled this unit, unfortunately.  I am also totally not a programmer, just a musician trying to do cool things with this great system.

thank you!

apossibleworld, 2020-09-11:
Can anybody help me troubleshoot this?  The two previous owners don't know who built it, but say it used to work fine.  Are there older versions of the hardware that don't work with firmware 2.0.1?

infovore, 2020-12-22:
There _is_ an older version of the hardware that needs a special compilation flag set for it. The obvious way to identify it is from the screws that hold the panel to the board: if they are all at the 'edge' of the board - two on each side towards the CV jacks, and one at the bottom corners - that's a 1.2X board. If the screws are laid out in two "X"s, then you've got a 1.3x board.

If it's a 1.2X board, reply here, and I can compile you a 2.0X firmware that'll work for you.

infovore, 2020-12-30:
(This was related to it being a 1.2x board! So I'm closing the issue.)

closed 2020-12-30 by infovore