16n-faderbank/16n #47: Fader curve issues.

kmtksk, 2020-09-17:
I made 16n and everything is working perfectly.
However, I checked the MIDI values and when I put the fader in the middle, the value is around 50 instead of 63.
Is this the correct behavior?

infovore, 2020-12-22:
Not really. It should be around 63.

Do you definitely have linear and not log faders in your 16n? Is the calibration set correctly? There are [configuration instructions](https://github.com/16n-faderbank/16n/wiki/Firmware:-installation-instructions) on the wiki.

infovore, 2020-12-22:
(If, after configuration, you have a non-linear fadercurve, it might be you have a 16n with log faders. Did you build it yourself? If not, who did? Can you check with them?)