16n-faderbank/16n #55: Can’t connect my fader bank to the editor

Photozik, 2021-03-17:
Hi,
I tried to connect my faderbank to the editor via Chrome, but it seems it can’t be recognized.
The editor sent this message : "It looks like a 16n is trying to connect, but may have a corrupt memory », and this : "Click to reset your 16n's EEPROM to factory defaults ». I clicked but nothing happens.
Is there a way to resolve this issue ?
Thank you very much.

infovore, 2021-03-23:
Sure, let's go over some diagnostics:

* what version firmware is your 16n running?
* was the Teensy on it used for anything else beforehand?
* did you assemble the 16n yourself?

Photozik, 2021-03-23:
Thanks Tom.
I didn't assemble the 16n myself. It's a second hand. It was originally made by Big T Music in UK but I bought it in US.
So I don't know if the Teensy was used before but I guess no, as it is a "factory" like unit.
The version mentioned on the advertisement is 2.0.0.

infovore, 2021-03-23:
Ah, OK. You might need to upgrade the firmware to a more recent version, though this possibly gives me a clue to a bug on my end with the editor.

Latest firmware is **2.1.0**, and you can [download it here](https://github.com/16n-faderbank/16n/releases) - you just need `16n_v210.hex`. [Instructions for updating are here](https://github.com/16n-faderbank/16n/wiki/Firmware:-installation-instructions) - you just need to download a piece of software called Teensy Loader, hook up your 16n, and flash it - you don't need to compile any code or anything.

If you can upgrade it to 2.1.0, that'd help me continue debugging.

Photozik, 2021-03-23:
ok. Great !
I will do that and tell you.
Thanks.

Photozik, 2021-03-23:
It seems to work now ! The 16n interface is on.
I just see that some faders go from 0 to 122, and some from 0 to 123.

infovore, 2021-03-23:
That's great. OK, so what you've got is a minor calibration that needs doing, which is what the editor's for. Plug in your 16n and in the editor, click Edit Config, then go to the Device Options tab. The "maximum raw value" might be set to 8135. Make that number a bit smaller - start with about 8000, I'm guessing - and hit "Update Controller" top right. Then, see how far the CC values alter. Keep bringing the "maximum raw value" down a bit until you get full 0-127 travel.

Photozik, 2021-03-23:
Done ! Every fader at 127 !

infovore, 2021-03-23:
Great news. I'm going to close this thread now, enjoy your faders!

closed 2021-03-23 by infovore

Photozik, 2021-03-23:
Thanks a lot Tom !!!
Have a nice day.