16n-faderbank/16n #61: Are 100mm faders possible?

mikattack, 2021-11-06:
**NOTE:** Not a hardware guy, don't know the complexity of this request.

Having purchases a 16n from a maker and really liked it. When I asked whether having 100mm faders was possible, they said a PCB redesign would be necessary to implement that.

I'm opening this issue to see the viability of that.

The main desire here is more fine-grained control without a high amount of finger dexterity. With a longer throw range, the resulting values from the faders are more "spread out" and allow those of us with shaky or imprecise fingers to achieve smaller changes.

Hopefully that makes sense.

infovore, 2021-12-03:
Yep, your contact is right: it's very doable, but it requires a PCB redesign, because the faders are directly mounted to the board. You just need to redesign the PCB, moving everything around to accommodate the new 100mm fader parts, and then fab the board. It's not complex, because none of the circuits will change, just one part allocation, but it _is_ time-consuming, because of the number of things to move around.

16n, as it stands, uses 60mm faders. A 100mm version would be totally doable, but I'm not going to do that. It's exactly why I made the project open-source - so other people can remix the project to their needs. If the question is "is it possible", the answer is *yes*, definitely. If the question is "will you do it?" the answer is no. I'm going to close the issue because that feels like it answers the question?

closed 2021-12-03 by infovore

mikattack, 2021-12-03:
Answer the question nicely. And the response is appreciated!