16n-faderbank/16n #64: Using 16n faderbank on VCV rack2

godzillalaughs, 2022-05-29:
Hi-

thank you for reading.  I’m using the teensy based ,16n faderbank on VCV Rack2, 2015 MacBook Pro, 10.15 OS.  After assigning faders and moving sliders, VCV Rack2 crashes.  Any suggestions would be welcomed. 

Thank you in advance for help.

infovore, 2024-07-18:
No idea, I'm afraid. That sounds like a VCV issue; the faderbank is a very unexciting MIDI controller. Did you ever resolve this?

Dewb, 2024-11-12:
FYI, in general 16n should be expected to work just fine with VCV Rack 2.4+, either using the stock MIDI mapping utilities or using the [faderbank module](https://dewb.github.io/monome-rack/modules/faderbank/) in the monome package.

There may have been issues with specific past versions of VCV Rack, or specific modules used as mapping targets, would need more info to diagnose.