16n-faderbank/16n #65: 16n as sequencer

FantacciSimone, 2022-09-08:
One of the main characteristics of the 16n is that (obviously) all the output signals are simultaneous; this makes the 16n a controller and nothing more.

What would happen if the output signals (at least MIDI out) were sequential at regular clock intervals (from MIDI in)? It would become a real sequencer. The sliders could adjust 16 pitches by 16 steps ... or maybe 8 for the pitch and 8 for the velocity.

Daydream or could we really give new life to our beloved faderbank?

Pilzfuss, 2022-09-29:
This would be heavily awesome !

FantacciSimone, 2022-09-29:
In the meantime I have discovered this fantastic script for Norns, which is very close to what I dreamed of!
https://github.com/zjb-s/sempra/tree/main

ArthurGibert, 2024-05-08:
There is now a eurorack expander that allows that. Reads the fader values with i2c thru a stereo cable for CV values. 
https://random-works.co.uk/

infovore, 2024-07-18:
It's a nice feature suggestion, but a stretch to fit in the current firmware. Alternative firmware, taking midi clock, and emitting notes, would be possible - though the nQuencer, as linked above, does do this very well, including rest/tie handling I believe. It feels like a problem better solved by Norns, Max4Live, or other hardware, than being core featureset of the faderbank.

closed 2024-07-18 by infovore