16n-faderbank/16n #67: Teensy 3.1/3.2 shortage (Design for Teensy 4.x?)

NeusAap, 2023-06-08:
Hello,

Since PJRC put a large disclaimer on their website regarding the stock and supply of the NXP chip on Teensy 3.x's & Teensy LC (see: https://www.pjrc.com/store/teensy32.html), I was wondering if anyone has already thought about, or maybe even designed an alternative design using the Teensy 4.x platform.

I would guess this would be possible, since their pinouts and capabilities are almost identical, but I just had not found any information about this yet :)

If anyone has done such thing, please let me know. And if not and I'd decide to do this I'll also share it ofcourse.

Kind regards,
NeusAap

infovore, 2023-06-09:
Of course: a Teensy 4.0 fork is perfectly possible, and the licensing (and open-source nature) of this project mean anyone is free to work on it.

NeusAap, 2023-06-13:
Awesome. Sounds like a great idea! Probably way better choice to use RP2040 stock wise and price wise. Can't wait to your new design! Might have to make 2 versions then. I've found one Teensy 3.2 laying aroung my house, so I'll start building that one first :)

Thanks for the answer!

closed 2023-06-13 by NeusAap