16n-faderbank/16n #68: Teensy 3.2 -> 4.0 Migration

janczmok1970, 2023-11-27:
Hi, 
i wonder what functions are critical in the 3.2 that would stop to 4.0 migration ?
Background: I hardly can get a 3.2, but 4.0 seems pin compatible.

eagereyes, 2024-03-21:
I just built a 16n using a Teensy 4.0 and it works with basically no modifications. I just commented out all the I2C code because the i2c_t3 library doesn't exist on 4.0 and I don't need I2C anyway.

That aside, it works just fine if I set the maximum raw value in the config utility to 4090. I bet there's a setting somewhere to get higher resolution, but since we're only sending 7-bit values anyway, it doesn't matter.

eagereyes, 2024-06-02:
Hi, I've forked the repo and uploaded my modified version of the firmware here: https://github.com/eagereyes/16n-for-Teensy-4.0 It also includes a compiled .hex file for use on Teensy 4.0. Let me know if it works for you!

Cy-Rot, 2024-06-16:
Hi, the hex file works fine, but I can't compile the source code.
I was on an older version of the arduino Ide, wich raised a lot of errors with midi.h.

I upgraded to 2.3.2 now but there are still a lot of warnings concerning sign and unsigned values and one ( as I assume) one error, which prohibits the teensy from connecting as a usb midi interface to my computer:
********************************************************************************************************************************
__In file included from C:\Users\samoz\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\libraries\MIDI\src/MIDI.h:307,
from C:\_DATEN\05_THO\CyFader\Cy_16n-for-Teensy-4.0main\firmware\16n_faderbank_firmware\16n_faderbank_firmware.ino:22:
C:\Users\samoz\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\libraries\MIDI\src/MIDI.hpp:492:74: note:   initializing argument 1 of 'void midi::MidiInterface<Transport, _Settings, _Platform>::sendRealTime(midi::MidiType) [with Transport = midi::SerialMIDI<HardwareSerial>; _Settings = midi::DefaultSettings; _Platform = midi::DefaultPlatform]'
  492 | void MidiInterface<Transport, Settings, Platform>::sendRealTime(MidiType inType)
      |                                                                 ~~~~~~~~~^~~~~~_
*********************************************************************************************************************************

Maybe you've have an idea how I could solve this?

infovore, 2024-07-18:
I believe 4.0 is pin compatible; the firmware will require recompiling. The hex file linked above by @eagereyes might be your best bet. Meanwhile, I've also [made a branch that reverts to using Wire.h](https://github.com/16n-faderbank/16n/tree/teensy4.0_spiking); the main advantage of that would be that Teensy 4 boards could be used along with I2C.

I do not, however, have a Teensy 4.0 to test with.

miotislucifugis, 2025-04-13:
For my port of the telex teletype expander code to teensy 4.0   [https://github.com/miotislucifugis/Telex_teensy4?tab=readme-ov-file](url) it turned out the fix was stupidly simple: just swap the I2C library for one that is compatible with teensy 4.x!    that was pretty much it, and was essentially a drop in replacement.   I first tried with Wire, and while it worked, it wasnt fast enough and would eventually cause I2c glitches where devices would stop responding.    I ultimately used this library:  https://github.com/Richard-Gemmell/teensy4_i2c .    Like I2C t3,  it is wire-compatible and pretty much uses the same callbacks.   Only the setup, needs to be modified slightly, as I2C t3 has a few extra variables.  ( teensy4_i2c has the same setup variables as wire)