16n-faderbank/16n #69: N16 not working

DanLeuca, 2024-09-24:
I have bought a N16 built by someone else but it does not work... when moving the faders the LED blinks but the computer would not 'see' the faderbank at all, so I cannot load the config files...

infovore, 2024-10-03:
Immediate question: is the USB micro-type-B cable you're using a cable that transmits data? A lot of use have USB micro cables lying around from, eg, mobile phone chargers that only have the power lines wired.

The LED flashing just indicates that the device is emitting MIDI; it's no indication of any actual connection.

Also - does it even appear over USB in some kind of System Information tool (I forget where you'd find this on Windows, on MacOS it's Apple Menu -> Hold Option -> System Information, and see if it's on one of the USB buses). If it's not showing up there, my first thought is: dud cable.

DanLeuca, 2024-10-04:
Thank you very much for your help. Do you know the exact name for the cable, have you got a link to some good product/supplier! Thanks in advance. Regards,Dan Sent from my iPhone

infovore, 2024-10-07:
Honestly, as long as the cable description mentions data transfer, you're good. Eg, an Amazon Basics one should be absolutely fine. Just choose the correct connector for your device - USB-A or USB-C, with USB micro-B at the other end, and you should be good. You don't need to spend lots (and of course, it might _not_ be the cable. But a new cable is cheaper than a cable tester...)