16n-faderbank/16n #70: 2.11 LC fader death

Thomasbru, 2024-09-27:
Updating to 2.11 breaks midi on some faders for me. On immediate update, fader 14-16 does not send midi, after a reboot, fader 13 is dead. I have not checked voltages, but the midi light does not blink for these faders as it does for the rest.

Downgrading to 2.10 solves this issue.

I have an AtoV project rework w/LC.

infovore, 2024-10-03:
OK, that sounds like an issue with the mux moving too quickly or something. I'll have a look at the code changes between 2.10 and 2.11.

infovore, 2024-10-03:
[You can see the changes here](https://github.com/16n-faderbank/16n/compare/9fc2dc6646f223f9a9f4dbc303892142f3c391f1..HEAD).

Two immediate thoughts:

- is the `delayMicroseconds` on line 344 causing an issue? (I don't think so, it's just a no-op)
- my other thought is the change on line 353, using `hasChanged`; it's possible the analogRead value hasn't settled after the mux changes channel, and that means it'll look like it's never updated. One way to solve this: _repeat_ line 353 two times after it - ie, read the analog channel into temp multiple times; that will give it time to settle to the new value coming through the mux.

I can probably supply you a build if you can't compile your own; those are my first thoguhts.

infovore, 2024-10-03:
No worries. I'd be interested to know the fix; I'd ideally like to merge it back into upstream.

I'm going to close the issue for now though, and we'll see if we can resolve this in due course.

closed 2024-10-03 by infovore