16n-faderbank/16n #74: Tried flashing with loader, now board no longer lights up and controller is not recognized

shocktropics, 2025-04-19:
I recently picked up a 16n that, while recognized by my DAW with corresponding orange LED, was not recognized by the online Editor. I tried following instructions for flashing the latest firmware, which worked fine with the Teensy Loader and the hex file etc.

The thing is, after it finished loading and rebooting, the 16n no longer has an orange LED no matter what I do, and it doesnt seem to be on at all.

Did I brick it somehow? Please help! These things are integral to my work!

shocktropics, 2025-04-19:
I have gone through the teensy loader firmware blinking a couple additional times. The loader seems to recognize when I push the little reset button on the board, and flashes firmware, automatically reboots, etc. Seems to be all good...but the 16n itself still has no LED activity, and is not recognized by the editor or the DAW or anything...!

shocktropics, 2025-04-19:
UPDATE: I seem to have gotten it sorted out! Through trial and error with the teensy loader app, I found that the board that is an "LC", and I found a corresponding firmware, flashed it again, and now everything seems to be good.

shocktropics, 2025-04-19:
another update: while the 16n now seems to be working fine within the DAW, it is still not recognized by the web editor...hmmm

infovore, 2025-04-21:
We're having issues with 2.2.0 with LC boards.

https://github.com/16n-faderbank/16n/releases/tag/v2.1.1

2.1.1 should remain compatible with LC boards - there are links to LC binaries there. You haven't bricked your 16n.

shocktropics, 2025-04-22:
thank you for the response. Through trial and error I indeed discovered that the particular 16n was an LC board, and subsequently I flashed the correct 2.1.1 firmware and it works fine.

Does its non-recognition thru the web editor correspond to issues with the LC board as well?